wgcv/RewriteM600 #1: octoprint install fails - 404 error

aft7, 2020-06-25:
install attempt gives :

Installing plugin "Rewrite M600 - For M600 Unsupported" from https://github.com/wgcv/RewriteM600t/archive/master.zip.../home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/wgcv/RewriteM600t/archive/master.zip --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleCollecting https://github.com/wgcv/RewriteM600t/archive/master.zipHTTP error 404 while getting https://github.com/wgcv/RewriteM600t/archive/master.zipCould not install requirement https://github.com/wgcv/RewriteM600t/archive/master.zip because of error 404 Client Error: Not Found for url: https://github.com/wgcv/RewriteM600t/archive/master.zipCould not install requirement https://github.com/wgcv/RewriteM600t/archive/master.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/wgcv/RewriteM600t/archive/master.zip for URL https://github.com/wgcv/RewriteM600t/archive/master.zipError!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

plugin_pluginmanager_console.log  :
2020-06-25 00:35:12,028   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/wgcv/RewriteM600t/archive/master.zip --no-cache-dir
2020-06-25 00:35:15,057 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2020-06-25 00:35:15,057 > Collecting https://github.com/wgcv/RewriteM600t/archive/master.zip
2020-06-25 00:35:15,060 ! HTTP error 404 while getting https://github.com/wgcv/RewriteM600t/archive/master.zip
2020-06-25 00:35:15,061 ! Could not install requirement https://github.com/wgcv/RewriteM600t/archive/master.zip because of error 404 Client Error: Not Found for url: https://github.com/wgcv/RewriteM600t/archive/master.zip
2020-06-25 00:35:15,061 ! Could not install requirement https://github.com/wgcv/RewriteM600t/archive/master.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/wgcv/RewriteM600t/archive/master.zip for URL https://github.com/wgcv/RewriteM600t/archive/master.zip

Jipe95, 2020-06-25:
Me 2

Installing plugin "Rewrite M600 - For M600 Unsupported" from https://github.com/wgcv/RewriteM600t/archive/master.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/wgcv/RewriteM600t/archive/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/wgcv/RewriteM600t/archive/master.zip
HTTP error 404 while getting https://github.com/wgcv/RewriteM600t/archive/master.zip
Could not install requirement https://github.com/wgcv/RewriteM600t/archive/master.zip because of error 404 Client Error: Not Found for url: https://github.com/wgcv/RewriteM600t/archive/master.zip
Could not install requirement https://github.com/wgcv/RewriteM600t/archive/master.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/wgcv/RewriteM600t/archive/master.zip for URL https://github.com/wgcv/RewriteM600t/archive/master.zip
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

wgcv, 2020-06-25:
Please use  ```https://github.com/wgcv/RewriteM600/archive/master.zip``` for the installation, was update in the PR https://github.com/OctoPrint/plugins.octoprint.org/pull/538

Thanks for the report

wgcv, 2020-06-25:
Fixed with the merge

closed 2020-06-25 by wgcv