wgcv/RewriteM600 #4: Does not resume correctly

energiebrand, 2020-06-29:
When sending M600 on my Sidewinder x1 via Octoprint the Z-axis moves up 8-10cm (did not measure it exactly). When pressing RESUME in Octoprint the Z-axis does not move down but starts printing at that hight. I expect, that the Z-axis will move down the same way before continue printing.

wgcv, 2020-06-29:
I was checking and yes was an error I think that is solved with the commit:  https://github.com/wgcv/RewriteM600/pull/3/commits/595753cecbeab89d25fcb98fe59652f464e565e2

energiebrand, 2020-06-30:
Hi wgcv, thanks a lot for checking this issue! Any idea about a date for a corrected version? :-) I am not a SW developer, so implementing your fix directly in the code is not an option for me,...I think...

wgcv, 2020-07-01:
Update the plugin

PLA7IN, 2020-07-01:
Hello @wgcv. I had some strange issues with the 1.0.0 version like @energiebrand and some extra issues too. So I updated to the new 1.0.1 version in hope this fix everything. Now it currently just kinda pause so it stops in the process and doesn't move away so someone can change the material. I can resume printing but that's not a difference to just pause the print.

Haacker, 2020-07-02:
I just had the same issue on my SideWinder X1. The printer just stops and leaves the printhead on the print job, rather than moving it up 80mm as specified in the settings for the plugin.

wgcv, 2020-07-02:
Sorry was a bug solved in https://github.com/wgcv/RewriteM600/commit/f866a0f59a2d0ad4634af567cd6c23b63d0c2f2e

I tested the last version and work well, please if you have any issue or problem report!

PLA7IN, 2020-07-02:
Now the head moves perfectly (if it is supposed that he does some print movements in the air over the object and then goes down on the object). But now I have the problems I had before with 1.0.0: the extruder doesn't extrude/move anymore.

wgcv, 2020-07-02:
Sorry, what do you mean doesn't extrude? Did you press resume in Octoprint?

The normal process  is:
1) Printing
2) Stop
3) Get up 8cm and move to home
4) Wait until the user changes the filament ( you can use the display menu to change the filament or a plugin for octoprint)
5) From Octorpint press resume
6) Resume the print and print until finish or other M600.

PLA7IN, 2020-07-02:
Sorry that I wasn't excatly about this. Yeah, after sending a m600 over terminal it does all your discribed steps but in your point 6 (after resume press in octoprint) the printer did following:
He goes from his +80mm z Position over the started printing object in ca +80mm (so he just moves x and y) an started moving the head like he resume printing with this distance but without extracting material (this wouldnt be a problem. It just looked maybe unsupposed). After some head movements in the air, he stops, goes down on the object and moves the head like he want to resume printing but he doesn't extruder material.

With the Version 1.0.0 some das ago I printed an other object where retraction was active and the printer didn't extrude, but retract the material back and forth.

I hope I have explained it clearly enough :) thanks

wgcv, 2020-07-05:
That sounds very weird. 
What Octoprint version you have? Do you have Python 3 or Python 2?
What is your slicer? You are using any plugin with the slicer?
You can share the G-code (Few lines before and after the M600)

PLA7IN, 2020-07-06:
Octoprint: 1.4.0
Python: 2.7.12
Slicer: PrusaSlicer-2.2.0
The slicer doesn't have any additional Plugins.
I was sending m600 over the terminal the last time, but i sliced an object too that always acted similar:

Here is a bit code from this file:
```
M204 S1250
G1 F1200
G1 X154.019 Y146.706 E0.31732
M204 S1000
; stop printing object Shape-Cylinder id:0 copy 0

G1 F6240.000
G1 X150.206 Y143.507 E-0.90931
G1 E-0.09069 F1200.00000
G1 Z9.440 F7800.000
M600
; printing object Shape-Cylinder id:0 copy 0
G1 X150.752 Y143.516
G1 Z9.240
G1 E1.00000 F900.00000
M204 S800
G1 F1200
G1 X151.421 Y143.842 E0.03863
G1 X151.939 Y144.160 E0.03158
G1 X152.428 Y144.522 E0.03158
```

Questman01, 2020-07-08:
Hi i got the same issue like PLA7IN

When an M600 occours the Z is going up. X and Y are going to Park. After changing the Filament and klicking Resume in Octoprint an bigger Retract occours and the Head starts "Printing in the Air". After a few Movements Z is going down to the right position and starts to print (It Looks like it starts at the right XY and Z position). But in case of the big Retract before it took a while until Material is back in the Nozzel the the first few Movements are without printing anything.

The end result looks like the photo i have attached. Hope you can it it there.

[image]

Greetings Kai-Uwe

DanHudson15, 2020-07-08:
I am still experiencing the print head staying at the same Z height.  The head returns to the correct X and Y, but need to be manually set to the correct Z.

PLA7IN, 2020-07-09:
Here is the video I made with version 1.0.2. Sorry that I recorded it upright and with bad quality. I recorded it with my old Phone to show that I send m600 with my phone: [https://youtu.be/dFMHTNsRmWE](url)

DanHudson15, 2020-07-09:
I may be working, I seem to have a clogged nozzle right after the color change.

PLA7IN, 2020-07-09:
v1.0.5 acting similar. All movements without the one from my extruder are like expected. My extruder doesn't extrude and does only retractions movements.

DanHudson15, 2020-07-10:
OK, so it looks like I got a clog.  Second print is working as expected.

Enhancement request - Is there a way to add a beep?

wgcv, 2020-07-10:
That is weird! Sorry, if I didn't check this week because my was board burned so I'm waiting for the replacement but I think that I'm going to change with all cable and without the flex and without the board because that doesn't work very well. Soon as possible I will check. If someone can help with PR great! 
[image]

wgcv, 2020-07-10:
> OK, so it looks like I got a clog. Second print is working as expected. Enhancement request - Is there a way to add a beep?
> […](#)

Please open another PR for that!

DanHudson15, 2020-07-10:
PR opened!  I am willing to test/debug with you on this.

________________________________ From: Konstantin Wacker notifications@github.com<mailto:notifications@github.com> Sent: Thursday, July 9, 2020 4:30
PM To: wgcv/RewriteM600 RewriteM600@noreply.github.com<mailto:RewriteM600@noreply.github.com> Cc: HUDSON, DANIEL J. DHUDSON@hanover.com<mailto:

electricJP, 2020-07-10:
> v1.0.5 acting similar. All movements without the one from my extruder are like expected. My extruder doesn't extrude and does only retractions movements.

Hello, I'm also having this issue on my printer: Z will not go down back to right position, plus extruder does not extrude anything. However, after M600 resume, I can pause again the print and extrude manually from Control tab without any issue.

yecine06, 2020-08-14:
Hi i have a longer LK4 pro and i installed the plugin v1.0.5

When adding a pause, the printer stops correctly and goes up. But on restart it start printing without going down.

If you need any other informations please tell me.

yecine06, 2020-08-14:
I don't know at all how octoprint is working and the plugins also but something strange.

in the code for resuming you have the following like.
`if cmd and cmd == "resume":`

it's like it search for the resume command but in my terminal after resume i don't see that

here is what i have

`Recv:  T:187.14 /190.00 B:60.54 /60.00 @:40 B@:0
Changing monitoring state from "Paused" to "Resuming"
Recv:  T:188.18 /190.00 B:60.60 /60.00 @:23 B@:0
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N271 M105*35
Recv: ok T:188.26 /190.00 B:60.64 /60.00 @:22 B@:0
Send: N272 M83*31
Recv: ok
Send: N273 G1 E-0.8 F4500*39
Recv: ok
Send: N274 G1 E0.8 F4500*13
Recv: ok
Send: N275 G1 E0.8 F4500*12
Recv: ok
Send: N276 M82*26`

Asaiel65, 2020-11-08:
Hi.

Thx for this plugin. I experience the same pbm as mentionned up: when the print is resumed, the head moves on X/Y but not on Z, print 1 layer (in the air), then goes down Z axis.

How can I fix this plz?

tolson2000, 2020-11-09:
I have the same problem with the Z doing a layer in space before returning to correct Z on the next layer.

I looks like the M114 which gets the XYZ coordinates to save is happening after the relative Z movement has been done and using that Z plus actual Z as the Z returned in the return from pause.

I do not know where the M114 is generated in the code to put it at the beginning of the M600 plugin. I don't see it in the M600 plugin.

Asaiel65, 2020-11-22:
> I have the same problem with the Z doing a layer in space before returning to correct Z on the next layer.
>
> I looks like the M114 which gets the XYZ coordinates to save is happening after the [...]

@tolson2000 Seems this plugin is now discontinued. Have you managed to make it work? Any clue?

fredvol, 2021-07-06:
Same for me , when click on "resume" , the printing has a 8cm offset on Z .

Bupher, 2021-08-09:
Same boat here, the print continues 8 cm above the layer change.

DanHudson15, 2021-08-09:
I find the issue is in Cura, I have not checked the latest version, but it puts the M600 command in the wrong place.  If you manually edit the Gcode to move M600 to the layer count.  I think it is fixed in Cura 4.10!

PLA7IN, 2021-08-09:
I never used cura for my gcode but I had this problem too. So this issue was not cura only but Slic3r and Prusa Slicer too.

DanHudson15, 2021-08-09:
I know I need to use the resume on OctoPrint via web control to resume or it messes up for me.

Bupher, 2021-08-09:
I think I found a workaround. I just set the m600 offset distance to zero and did it all in the printer pause and restart gcode scripts.

r34220, 2021-11-04:
> Hi.
> 
> Thx for this plugin. I experience the same pbm as mentionned up: when the print is resumed, the head moves on X/Y but not on Z, print 1 layer (in the air), then goes down Z axis.
> 
> How can I fix this plz?

I am seeing the same on my sidewinder x1

Doprintityourself, 2022-02-18:
HI all,
M600 works fin, until after the resume Button hit print head moves correctly into xy z but for about 10-15 sec there is no material extruded. I made a small test sign and after being finished the border (as being printed the at first) came off.
So it seems that there is no extrusion or a retract that I didn't recognize after resume. 
As I extruded a good amount of filament manually during the filament change I am sure that there was instant filament available after resuming!.
Any Idea?

frogger42, 2022-03-03:
Same for me , when click on "resume" , the printing has a 8cm offset on Z .

Travja, 2022-03-26:
Well.. probably the oldest bug here, but I tackled it today. PR #12. I made a good number of modifications with how it functions, what the resume mechanics are, etc. I want to make it so it dynamically rewrites the M600 command as if it were put in straight from Cura with specific X/Y/Z offsets rather than just homing and calling it a day. Ultimately that'd be more of a nice feature than a required one as this does work currently. As I'm pretty sure that this project has been abandoned, I'll look into getting my rendition posted on the official OctoPrint plugin repository for you all, but for the time being you'd probably have to know a thing or two to manually install it. Maybe the PR will get approved and we'll all be cruising. Who knows.

Jeanne160, 2022-04-19:
> Hello, I'm also having this issue on my printer: Z will not go down back to right position, plus extruder does not extrude anything. However, after M600 resume, I can pause again the print and [...]

Hello all, 
Just found out this thread; I solved the problem!
I seems that whenever I send a M600 to my printer (Sidewinder x2), extrusion goes back to absolute mode (automatical M82).
Thus, after the pause, the first line commande ( G1 X143.094 Y152.546 E-.78089  for my specific gcode file) sends the filament out of the nozzle, since it is in absolute mode.
THE SOLUTION:
Based on that article: https://www.yirco.me/octoprint-pause-change-filament/
In the Octoprints parameters --> "Printer" --> "Gcode Scripts"
Add this in the "Before print job is resumed":
M83;
G1 E-0.8 F4500;
G1 E0.8 F4500;
G1 E0.8 F4500;

It basically switch extrusion to relative mode, and prepare the extruder for the next line.
After that, everything works perfectly fine.
Cheers
[image]

lupusworax, 2022-09-13:
Anycubic Chrion with stock firmware, M600 lifts and parks the nozzle correctly, but when pressing resume it starts printing with a 8cm offset on the Z axis.

Triptikal, 2022-09-29:
Hello all,

I was experiencing the same issues and, after a long research time, I found applicable fixes for me. And as far as I can see the M600Rewriter plug in was not the problem.
I know this topic is old now and maybe it was resolved in other topics but if I can help someone else 
My printer : Sidewinder X2
My slicer : Superslicer_2.4
I'm using Octoprint togheter with M600 Rewriter plug-in to interpreet the M600 command because the X2 does not supports it natively.

First problem : the fact that the extruder seems to go in a "reversed" way ( moving up the filament instead of moving down for extrusion )

Resolution : I first apply the solution to just add the M83 into octoprint's "Before print job is resumed
" g-codes but the problem was still there. So I decided to put the full "package" of information so I did put :

{% if pause_position.x is not none %}
**G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion**
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F4500
{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
{% endif %}

And it worked.

Second problem : the fact that the printhead was printing "in the air" for some seconds before suddenly going back to the correct layer.
Resolution : I found out, in the log, the G1 Z=XX code was going from Z=81.6 to Z=1.7 ( the famous 8 cm that need to be removed ) so that was the reason of the fact that suddenly the printhead was going to the correct layer. I checked a bit deeper the log and I found out that superslicer ( and I presume Pusaslicer too as it is the same base ) puts the M600 command in the middle of a layer change.So I moved the M600 command just before the layer change code and it worked also . The printhead was going back directly to the correct position after filament change as a correct Z position

I'm still trying to figure out how to fix this thing, within superslicer, without modifying my gcode everytime so if someone find a fix, I would be happy to have it :slight_smile:

GENERATED CODE BY SUPERSLICER :
;WIPE_START
G1 F6240
G1 X143.361 Y142.322 E-0.95 ; wipe and retract
;WIPE_END
G1 E-0.05 F4800 ; retract
; stop printing object Jetons_MATCHS(8).stl id:0 copy 0
M486 S-1
G1 Z1.32 F7800 ; move to next layer (4)
; custom gcode: layer_gcode
**;AFTER_LAYER_CHANGE
;1.32
; custom gcode end: layer_gcode
;COLOR_CHANGE,T0,#0ED51F
; custom gcode: color_change_gcode
M600
; custom gcode end: color_change_gcode**

MY MODIFIED CODE :

**;COLOR_CHANGE,T0,#0ED51F
; custom gcode: color_change_gcode
M600
; custom gcode end: color_change_gcode**

;LAYER_CHANGE
;Z:1.32
;HEIGHT:0.28
; custom gcode: before_layer_gcode
;BEFORE_LAYER_CHANGE
;1.32
G92 E0
; custom gcode end: before_layer_gcode

;WIPE_START
G1 F6240
G1 X143.361 Y142.322 E-0.95 ; wipe and retract
;WIPE_END
G1 E-0.05 F4800 ; retract
G1 Z1.32 F7800 ; move to next layer (4)
; custom gcode: layer_gcode
;AFTER_LAYER_CHANGE
;1.32
; custom gcode end: layer_gcode

lgallard, 2022-11-21:
Same here, in Cura it prints in the air for some secs and then contines and PrusaSlicer retracts the filament and can't print due to lack of filament. 

@Triptikal where this block should be?

```
{% if pause_position.x is not none %}
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F4500
{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
{% endif %}
```

In PrusaSlicer there's a section for the custom g-code when changing color (among others), should be there?

[image]

Herbert-Testmann, 2023-01-23:
> Hi.
> 
> Thx for this plugin. I experience the same pbm as mentionned up: when the print is resumed, the head moves on X/Y but not on Z, print 1 layer (in the air), then goes down Z axis.
> 
> How can I fix this plz?

same problem on Elegoo Neptune3
No Z move after M600.
print one layer
and then go down zu correct Z position for next layer.

If the Z offset is configured to"0" - it works.

DanHudson15, 2023-01-23:
I found I needed to reposition the M600 command in the Gcode to where the new Z height is issued for the next layer.  Cura puts the Layer marker in the wrong spot.

;MESH:NONMESH
;TIME_ELAPSED:4936.846211
M600 E30.00 U300.00 X0.00 Y0.00 ; Generated by FilamentChange plugin
;LAYER:19   <--  This is the layer marker that the plug in uses, but Cura puts in the wrong place in the code.
;TYPE:WALL-INNER
;MESH:Christmas Bell - Bill.stl
G0 F300 X137.557 Y130.369 Z3.28   <-- here is where the new layer height is set
G0 F9000 X137.593 Y130.367
G0 X137.983 Y137.986
G0 X137.983 Y139.985
G0 X144.665 Y166.229
G0 X144.665 Y168.228
G0 X149.989 Y169.22
G0 X150.061 Y177.137
G1 F1500 E3678.35395      <--   this is where the relative extruder is used.  Be sure to move the filament out as much as you want to push the new filament in.  I use “out” on the manual control twice, so I can purge by hitting  “in” twice.
G0 F9000 X149.741 Y177.427
< The layer marker was here.>
G1 F1800 X149.741 Y169.468 E3678.5869
G1 X144.431 Y169.468 E3678.74232
G1 X144.431 Y167.994 E3678.78546

This e-mail, including attachments, is intended for the exclusive use of the addressee and may contain proprietary, confidential or privileged information. If you are not the intended recipient, any dissemination, use, distribution or copying is strictly prohibited. If you have received this e-mail in error, please notify me via return e-mail and permanently delete the original and destroy all copies.