wgcv/RewriteM600 #10: Extruder don't resume

manubissanchez, 2021-11-28:
Hello,
I have an issue with this plugin.
At the resume it doesn't go down to the print and move to do a layer, then go down to the print and move for an other layer : all this with an extruder wich doesn't move at all.
I have a Sidewinder X1 Stock.
Thanks in advance.

grumpyniffler, 2022-01-21:
I have a similar problem.

**What works:** 
The printer stops -> goes in parking position -> I change filament via TFT -> I resume the print through Octoprint -> printer moves back to print

**What does not work:**
The printer then follows the given GCode paths but does not resume to extrude properly.
First I thought of a clogged nozzle, but when I start a new print, everything is fine. If I try M600 with different filament, I get the same result.´

Clearly there seems to be something broken in the code for resuming the print.
My printer also is a stock Sidewinder X1 V4.

Herbert-Testmann, 2023-01-23:
I have the same problem
Nozzle move to X/Y Position 0,0 and Z with offset in plugin config.
If I press resume on Octoprint WEb UI the printer go back to correct X/Y Position, but not to Z. The Nozzle goes not down to object. Then print a Layer in hight Z position und if the layer ends the printer start the next layer on correct Z position.
So a full or half layer is missing in the object.
Printer ist Elegoo Neptune3.
A workaround is ... set the Z offset in config on "0"